Miceuz/PlantWateringAlarm #2: Software License

losttime, 2016-04-09:
The CERN OHL states that it "does not apply to software, firmware, or code loaded into programmable devices".

How would feel about specifying a license for the code you've provided? [MIT](https://opensource.org/licenses/MIT) is a favorite of mine.

andrew-sz, 2021-04-15:
This issue was opened long ago.   I concur, I cannot find a software license.   Anyone still maintaining this repo?

Miceuz, 2021-04-15:
Yes I do. The whole thing is licenced under CERN OHL licence.

Miceuz, 2021-04-15:
Fw - I dunno, never cared to choose. Say, GPL2?

andrew-sz, 2021-04-19:
Ok, thanks for the reply!     It is entirely up to you but I think that GPL2 has the same intent as Cern's OHL v1.1.   btw this is great project!   And thanks for sharing the code and hardware designs!

pojntfx, 2021-05-04:
I'd much appreciate GPLv2 as well! I'm using a Chirp! in https://github.com/pojntfx/growlapse (which is AGPLv3) and this would clear up the license situation :)