Miceuz/PlantWateringAlarm #3: Operation with lithium cells

dewi-ny-je, 2017-02-27:
Since I have multiple plants next to each other I was thinking about powering them with a rechargeable lithium cell.

Would Chirp work with 4.1-3.3 V input? What is the tested operating range?

Miceuz, 2017-02-27:
Tested operating range is 3.3 to 5V. The circuit is really sensitive to voltage swings, so better add a low drop out regulator to keep it at 3.3V. You can get one in sot-23-5 package with enable pin which you could use to power down all the sensors. Put your I2C pullup resistors on the same regulated 3.3V so that they can get shut down also.

dewi-ny-je, 2017-02-27:
Thanks for the info. I'm buying some MCP1700-3.0 that should have 1.6 uA quiescent current, to keep things simple.

closed 2020-02-10 by Miceuz