Miceuz/PlantWateringAlarm #4: Disable BOD not working

cimba007, 2017-07-11:
> In order to disable BOD during sleep (see Table 7-1 on page 33) the BODS bit must be written to
> logic one. This is controlled by a timed sequence and the enable bit, BODSE in MCUCR. First,both BODS and BODSE must be set to one. Second, within four clock cycles, BODS must be
> set to one and BODSE must be set to zero.

I think the current implementation violates the "4 clock cycle rule". OR ing the bits takes too much time. This results in an power down current consumption of ~27µA (BOD is active if not disabled by fuses).

I propose the following implementation of the sleep function.

```
void inline sleep() {
    set_sleep_mode(SLEEP_MODE_PWR_DOWN);
    sleep_enable();
    //MCUCR |= _BV(BODS) | _BV(BODSE);    //disable brownout detection during sleep
    //MCUCR &=~ _BV(BODSE);
    uint8_t mcucr1 = MCUCR | _BV(BODS) | _BV(BODSE);
    uint8_t mcucr2 = mcucr1 & ~_BV(BODSE);
    MCUCR = mcucr1;
    MCUCR = mcucr2;
    sleep_cpu();
    sleep_disable();
}
```

There might still be room for improvement (https://github.com/LowPowerLab/LowPower/blob/master/LowPower.cpp#L41) but this solution works for me. 

Now my chirp consumes only 6µA instead of 27µA during power down.

Miceuz, 2017-07-12:
Thank you for the catch! I will address it in my next batch.

Miceuz, 2017-07-12:
Now when I think about this more - disabling BOD might be a bad idea resulting in a corrupted flash, don't you think?

cimba007, 2017-07-12:
During active operation more current will be consumed resulting in a potentially lower voltage of the coin cell. Disabling BOD during sleep (a period with low current consumption) should always result in a higher battery voltage.

So disabling BOD during sleep should be safe. After sleep wakeup BOD will be enabled and if the voltage is too low should result in a reset condition.

The only think you should thing about is why the CPU is fused to 8mhz as it ius not save for 3V, better run all the time with 1mhz which allows 1,8Volt of operational current.

cimba007, 2017-07-12:
One addition:

    _delay_ms(1);
    getADC1();
    _delay_ms(1000);

WIll take 4 Seconds instead of the intended 1 second. Can you tell me more about the reason for this delay? I guess you ran some test ;-)