Miceuz/PlantWateringAlarm #7: Are External I2C Pullup Resistors Required?

vicatcu, 2018-01-19:
Hey, sorry this is not really an issue per say, except maybe a documentation issue. Are external pull-up resistors required if you are using the stand-alone sensor version to interface with an Arduino over I2C? 

I don't see any in the schematics, but maybe I missed something. If indeed they aren't there, might not be a horrible idea to put pads on the PCB to (optionally) populate them.

Miceuz, 2018-01-19:
Idea being that there could be other devices on the bus and bus design is a higher level task, I think master should provide pullups. Say, its totally a designer decision on the size of pullup resistors.

vicatcu, 2018-01-19:
Yup, I dig it, and I agree in general. The question is really can you hook it up (as the only device on the I2C bus) to an Arduino Uno directly and communicate successfully? For example using [this library](https://github.com/Apollon77/I2CSoilMoistureSensor) or the example code on [your page](https://wemakethings.net/chirp/). It wouldn't hurt to put a reminder to developers that these external resistors _might_ be necessary in a given application.

Miceuz, 2018-01-19:
Yes, you can hook it up directly to Arduino. Their internal pullups by default are on and are enough to test.

closed 2020-02-10 by Miceuz