Miceuz/PlantWateringAlarm #12: Switch of read out registers when resetting

itofficeeu, 2019-05-04:
When I have the sensor hocked up to I2C and I use the reset on the top of the sensor, can it happen that the read out registers switches. Moisture then is read from the register for light and the other way around.

I do not know if this depends on the producer. I do not now if this can be fixed by software. I can reveal that my hardware does not respond with temperature data. So maybe is it only the product I have in hand? - Be aware from where you buy, if the temperature sensing is important. Try out a small quantity first.

One should be prepared for sanitizing the switched data afterwards or maybe even build in an automatic sanitation.

DrBabbage, 2022-07-18:
I might have the same issue. Have you found a fix?

itofficeeu, 2022-07-18:
Nope. Dropped working with the sensor.

Miceuz, 2022-07-19:
Chirp is not ment for proper installations, just play around with a pot nearby. For proper performance look into standalone I2C sensor

closed 2022-12-15 by Miceuz