Miceuz/PlantWateringAlarm #13: Sensor damaged by soil

git-torrent, 2019-06-09:
Hello,

my sensor got damaged by soil (paint peeled off). Readings go from 200 to 2000, I can't use it anymore. I would provide more details upon request (e.g. picture). Thank you

Miceuz, 2019-06-10:
Hello,

Can you please give more information:
 * What is the version written on the sensor?
 * How long did it take before failure?

Miceuz, 2019-06-10:
Also can you plese make a picture of the damage on the other side of the sensor?

git-torrent, 2019-06-12:
Hello, for sure I am sending both sides. In two separate emails

[image: 20190611_195831.jpg]
All the best

Pavol Ostertag

git-torrent, 2019-06-12:
I am sending both sides, for sure.
[image]
[image]

Miceuz, 2019-06-12:
How long did it take before failure?
And can you please elaborate if the sensor was re-inserted into soil a lot?

git-torrent, 2019-06-12:
The sensor was inserted into soil in October 2018. Back then, values were ca 420 for freshly watered soil and ca 160 for dry soil (I pulled the sensor out). So I calibrated it to range 150 - 400. During winter the sensor remained in the soil. When I started watering again (in March), the readings were always 100% - fully wet. Only when heats started and soil became dry, I started to wonder why the sensor still reads "fully wet". That was the first time I pulled it out. I cleaned the soil from the tip and the result is in the pictures.

The soil is for blackcurrant so it is maybe little bit acidic (Ph6) when fertilized.

Miceuz, 2019-06-14:
Hi, 
Thank you for your report. Yes, sensors are not totally bullet proof. Especially when it comes to freezing ground. Can you ship me the sensor for examination? I would ship you a replacement. Please contact me via email - mic at wemakethings dot net

closed 2020-02-10 by Miceuz