Miceuz/PlantWateringAlarm #14: Is it possible to use 2 or more sensors in only 1 i2c bus?

thedonalddon, 2019-07-22:
I have three plants, and I bought 3 sensors for each of them. I'm using an ESP8266 as I2C master and a Raspberry Pi as the controller. 
But on ESP8266 I could use only 2 GPIOs as SCL and SDA. Is it possible to connect 3 chirps at the same time using the same i2c bus? 
(sorry the question might be silly, I'm kind of rookie about this..)

Miceuz, 2019-07-22:
You can hook up several sensors on the same SCL, SDA lines. Before doing that you have to hook up a single sensor to the bus one by one and set different addresses for each sensor. Don't forget to add pullup resistors if they are not on the esp8266 board yet. 

Also if you experience communication stability problems, try lowering pullup values and/or communication speed. 

Also it seems you could use a basic tutorial on I2C bus - Sparkfun has a pretty good one that's easy to find.

closed 2019-07-22 by Miceuz

thedonalddon, 2019-07-23:
The tutorial is really helpful! Should've read it earlier. Thanks!