Miceuz/PlantWateringAlarm #16: Do recent sensors ship with the firmware from this repo?

Staars, 2019-09-29:
Hello,

I somehow bricked the I2C-communication to my sensor, probably in the process of assigning a new address, and before I try to reflash it, I want to ask the question above.

The version in /src is 2 years old and does not contain a version number. Besides it still does not include this PR: https://github.com/Miceuz/PlantWateringAlarm/pull/11

Thanks a lot and best regards

Christian

tmm1, 2020-02-08:
Did you end up reflashing to fix i2c?

Staars, 2020-02-08:
Yes and it fixed the problem.
Sidenote: I had bought 2 cheap chinese replicas dead-on-arrival and could bring one to life by flashing it.

Miceuz, 2020-02-10:
Sensors are shipped with the firmware from branch board-update-2.4.5 
In general, all chirps I sell are HW version 2.4.5 with the updated firmware. Regarding clones form Aliexpress - I have no idea and no control over it.

closed 2020-02-10 by Miceuz