Miceuz/PlantWateringAlarm #17: re-programming

palitu, 2021-12-14:
I wanted to re-flash the chirp, to learn a bit more about how this all works.

I have looked through the repo, and as far as i can tell the standalone code lives in `PlantWateringAlarm/src/` and to compile and upload to the device, i simply run `sudo make program`.

this will compile the code, set the fuses and upload the firmware.  

I am not 100% sure on what is in the programmer directory, but i assume it is to use a microcontroller as programmer for ISP headers?

Can someone confirm this?

cheers

Miceuz, 2021-12-14:
Hi there,

Programmer directory contains files for programming jig I used to use at some point in time, it has no relevance for the project itself nowadays. I should add a readme file there.

DrBabbage, 2022-07-18:
`apt-get install gcc-avr avr-libc -y`

then 

`make hex`

DrBabbage, 2022-07-18:
have you managed to get it working?

Miceuz, 2022-07-19:
Run `make hex` to compile the firmware. Run `make fuse` to flash fuses. Run `make flash` to flash the firmware. You need `avrdude` and `USBasp` programmer device to flash out of the box. You can use arduino board as an ISP programmer, but you will have to update `makefile` to make it work.

closed 2022-07-19 by Miceuz